NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #1: Low fps from Arducam-IMX219

kaganGH, 2021-11-07:
Hello,

I am using a Jetson Xavier NX development board for my project. The power mode I set is "Mode 20W 6Core". I am using Arducam-IMX219, which are natively supported. I have built and run the docker successfully and then cloned and built "isaac_ros_argus_camera" without any error. When camera is seen at `/dev/video0`, i run the following command to launch the node:

`ros2 run isaac_ros_argus_camera_mono isaac_ros_argus_camera_mono --ros-args -p device:=0 -p sensor:=4 -p output_encoding:="mono8"` 

which corresponds to 1280x720 resolution for my camera and it should run at 60fps according to `v4l2-ctl --dev?dev/video0 --list-format-ext`

However, I can get at most 24-26 fps when i check the topic `/image_raw.` What could be the possible reason? Do I need to activate any GPU related settings (CUDA)?  Can the nvpmodel be the issue? Are the other parameters that I can set for this node? Thank you in advance.

kaganGH, 2021-11-07:
Additional info:

If I run the gscam2 node with:

`export GSCAM_CONFIG="nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, format=BGRx ! videoconvert"`
`ros2 launch gscam node_param_launch.py`

I am able to obtain nearly 60 fps. But it would be nice to reach this rate with GPU implementation as well since I need CPU for other tasks.

glonv, 2021-11-09:
Please see also [link]( https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_image_pipeline/issues/2) to maximize Jetson NX performance.

kaganGH, 2021-11-13:
Hello again,

Here are my output when I run first `sudo /usr/bin/jetson_clocks` and `sudo /usr/bin/jetson_clocks --show`

```
SOC family:tegra194  Machine:NVIDIA Jetson Xavier NX Developer Kit
Online CPUs: 0-5
cpu0: Online=1 Governor=schedutil MinFreq=1907200 MaxFreq=1907200 CurrentFreq=1907200 IdleStates: C1=0 c6=0 
cpu1: Online=1 Governor=schedutil MinFreq=1907200 MaxFreq=1907200 CurrentFreq=1907200 IdleStates: C1=0 c6=0 
cpu2: Online=1 Governor=schedutil MinFreq=1907200 MaxFreq=1907200 CurrentFreq=1907200 IdleStates: C1=0 c6=0 
cpu3: Online=1 Governor=schedutil MinFreq=1907200 MaxFreq=1907200 CurrentFreq=1907200 IdleStates: C1=0 c6=0 
cpu4: Online=1 Governor=schedutil MinFreq=1907200 MaxFreq=1907200 CurrentFreq=1907200 IdleStates: C1=0 c6=0 
cpu5: Online=1 Governor=schedutil MinFreq=1907200 MaxFreq=1907200 CurrentFreq=1907200 IdleStates: C1=0 c6=0 
GPU MinFreq=1109250000 MaxFreq=1109250000 CurrentFreq=1109250000
EMC MinFreq=204000000 MaxFreq=1866000000 CurrentFreq=1866000000 FreqOverride=1
Fan: PWM=0
NV Power Mode: MODE_20W_6CORE_MAX
```

I think I already use the max power from cpu and gpu's. However I am still getting around 20fps with "isaac_ros_argus_camera". I have found two things strange. **First, while the camera package running, the gpu is nearly not used and one of the cpu's are 100% this is a shot from jtop**:

[image]

**Second, I can get more fps with gscam_node. I think this is fully cpu implementation. I suprised that hardware accelerated version (argus_cam) is slower than this.** Still I am not sure if my hardware settings are correct. 

Do you have any more suggestions? How much fps/resolution you can get from your setup? Thank you.

kaganGH, 2021-11-13:
Note: I know that this power settings might not be recommended for a healthy operation (due to temp increase) but I set this to see how much fps I can get.

glonv, 2021-11-17:
I can reproduce the low fps with IMX219 on Jetson NX. With 60fps@720p sensor mode, outputting messages run less than 20fps. I will try to profile the latency and see what could be the delay.

hemalshahNV, 2022-09-19:
@glonv could you reproduce this with more recent version of Isaac ROS?

closed 2022-10-24 by jaiveersinghNV

KnechtNoobrecht, 2024-02-08:
Hello, not sure if relevant, but i can reproduce this on a Jetson Nano Devkit running ROS Noetic and gscam 1.0.1.
My GSCAM_CONFIG export is identical to kaganerunsal's. Though i get desired performance using `gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=(int)3264, height=(int)2464, framerate=(fraction)21/1' ! nvv4l2h264enc enable-lossless=true profile=4 ! h264parse ! mp4mux ! filesink location=filename.mp4 -e` **in a preview window outside of ROS and gscam**. I'd be very happy about a solution for the mediocre performance in ROS.

rodri-go-fr, 2025-07-17:
Even with the last version of jetpack running on Isaac Dev Container is even worse im only getting the first frame and then it frezees