NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #2: Arducam IMX219 Camera Info Message

kaganGH, 2021-11-07:
Hello,

I am using a Jetson Xavier NX development board for my project. I have Arducam-IMX219, which are natively supported. Nearly for all my projects with ROS and camera, I was using the camera calibration tool  (http://wiki.ros.org/camera_calibration) to obtain camera intrinsic parameters, save them as .yaml or .ini, the give this as a parameter to ROS camera driver such as usb_cam (http://wiki.ros.org/usb_cam) to successfully publish camera_info message.  As I understand isaac_ros_argus_camera node expects these parameters  stored in EEPROM of camera which I think is not possible for every brand. Is there way to run this node with this kind of parameter file (.yaml or .ini)? Thanks a lot in advance.

glonv, 2021-11-09:
Argus [Calibration data API](https://docs.nvidia.com/jetson/l4t-multimedia/classArgus_1_1Ext_1_1ISyncSensorCalibrationData.html) is public and not limited by any type of storage or brand. isaac_ros_argus_camera currently doesn't support calibration file input but this will be considered in the future release.

hemalshahNV, 2021-11-23:
With Isaac ROS EA2.1, we were able to get in a camera_info_url parameter to the monocular nodes at least to solve your issue (reads from a yaml file and publishes the CameraInfo rather than read from EEPROM only). We'll add the same for stereo node soon. Let us know if it works for you!

kaganGH, 2021-12-06:
It is a good news. Thank you. I will try and inform you.

closed 2022-07-04 by gordongrigor