NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #8: Error Could not find Argus/Argus.h

kyuhyong, 2022-05-10:
I am running ROS2-foxy on jetson nano.
I installed everything required to build this package however I am still getting these error(s)
```
--- stderr: isaac_ros_argus_camera_mono                       
/home/r1mini/ros2_ws/src/isaac_ros_argus_camera/isaac_ros_argus_camera_mono/src/argus_camera_mono_node.cpp:17:10: fatal error: Argus/Argus.h: No such file or directory
   17 | #include <Argus/Argus.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/monocular_node.dir/build.make:63: CMakeFiles/monocular_node.dir/src/argus_camera_mono_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/monocular_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< isaac_ros_argus_camera_mono [3.80s, exited with code 2]
Aborted  <<< opencv_tests [14.4s]                                         
Aborted  <<< isaac_ros_argus_camera_stereo [23.3s]
```
Found a topic regarding this issue and tried to put 'set(TegraMM_ROOT $ENV{HOME}/jetson_multimedia_api)' on top of CMageLists.txt but no luck. 
https://forums.developer.nvidia.com/t/error-when-using-argus-camera/180137

How to solve this issue?
Thanks,

hemalshahNV, 2022-07-13:
Could you confirm which Jetpack version you're running on the Nano? You'll need Jetpack 4.6.1 if you're running the last Foxy-compatible Isaac ROS release.

closed 2022-08-01 by hemalshahNV

KnechtNoobrecht, 2023-11-20:
> Could you confirm which Jetpack version you're running on the Nano? You'll need Jetpack 4.6.1 if you're running the last Foxy-compatible Isaac ROS release.

Hi. I am having the same problem on a Jetson Nano running Jetpack 4.6.4. How do i solve this?