NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #15: How to set the FPS and Resolution of camera?

lymnxn, 2022-12-02:
Hello,
I'm using a Jetson Xavier NX and a IMX219 camera for developent. I want to get high FPS video so I need to set camera output as 1280x720@60fps, I can not find any tutorials about it.

jaiveersinghNV, 2022-12-06:
You'll need to find which profiles are supported by your IMX219 sensor, by using a command like `v4l2-ctl`. 

This should output the supported resolutions and FPS.

Then, edit [these lines](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera/blob/main/isaac_ros_argus_camera/config/argus_camera_mono_node.yaml#L69-L72) of the Argus configuration to use the mode and FPS values from the previous step.

lymnxn, 2022-12-06:
I have another Question, why I have set the camera calibration data with a valid url, I still get the warning:
 [component_container_mt-1] 2022-12-06 07:50:10.436 WARN  extensions/hawk/argus_camera.cpp@677: Failed to get calibration data from Argus!
**Here is the full log**:
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [66891]
[component_container_mt-1] [INFO] [1670313006.428758395] [argus_mono_container]: Load Library: /workspaces/isaac_ros-dev/install/isaac_ros_argus_camera/lib/libmono_node.so
[component_container_mt-1] [INFO] [1670313006.961008126] [NitrosContext]: [NitrosContext] Creating a new shared context
[component_container_mt-1] [INFO] [1670313006.961802563] [argus_mono_container]: Found class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::argus::ArgusMonoNode>
[component_container_mt-1] [INFO] [1670313006.961943940] [argus_mono_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::argus::ArgusMonoNode>
[component_container_mt-1] [INFO] [1670313006.984610955] [argus_mono]: [NitrosNode] Initializing NitrosNode
[component_container_mt-1] [INFO] [1670313007.142159958] [argus_mono]: camera calibration URL: package://viconros2/config/csi_cam.ini
[component_container_mt-1] [INFO] [1670313007.146437647] [argus_mono]: [ArgusMonoNode] Loaded camera info from "package://viconros2/config/csi_cam.ini"
[component_container_mt-1] [INFO] [1670313007.146604880] [argus_mono]: [NitrosNode] Starting NitrosNode
[component_container_mt-1] [INFO] [1670313007.146663377] [argus_mono]: [NitrosNode] Loading built-in preset extension specs
[component_container_mt-1] [INFO] [1670313007.158875609] [argus_mono]: [NitrosNode] Loading built-in extension specs
[component_container_mt-1] [INFO] [1670313007.159147995] [argus_mono]: [NitrosNode] Loading preset extension specs
[component_container_mt-1] [INFO] [1670313007.162802513] [argus_mono]: [NitrosNode] Loading extension specs
[component_container_mt-1] [INFO] [1670313007.162935058] [argus_mono]: [NitrosNode] Loading generator rules
[component_container_mt-1] [INFO] [1670313007.164034744] [argus_mono]: [NitrosNode] Loading extensions
[component_container_mt-1] [INFO] [1670313007.165583393] [argus_mono]: [NitrosContext] Loading extension: gxf/std/libgxf_std.so
[component_container_mt-1] [INFO] [1670313007.174391062] [argus_mono]: [NitrosContext] Loading extension: gxf/cuda/libgxf_cuda.so
[component_container_mt-1] [INFO] [1670313007.180597723] [argus_mono]: [NitrosContext] Loading extension: gxf/serialization/libgxf_serialization.so
[component_container_mt-1] [INFO] [1670313007.189290607] [argus_mono]: [NitrosContext] Loading extension: gxf/libgxf_sight.so
[component_container_mt-1] [INFO] [1670313007.196021783] [argus_mono]: [NitrosContext] Loading extension: gxf/libgxf_atlas.so
[component_container_mt-1] [INFO] [1670313007.204862731] [argus_mono]: [NitrosContext] Loading extension: gxf/libgxf_isaac_messages.so
[component_container_mt-1] [INFO] [1670313007.212249783] [argus_mono]: [NitrosContext] Loading extension: gxf/multimedia/libgxf_multimedia.so
[component_container_mt-1] [INFO] [1670313007.215058472] [argus_mono]: [NitrosContext] Loading extension: gxf/tensorops/libgxf_tensorops.so
[component_container_mt-1] [INFO] [1670313007.249297972] [argus_mono]: [NitrosContext] Loading extension: gxf/libgxf_argus.so
[component_container_mt-1] [INFO] [1670313007.676029571] [argus_mono]: [NitrosContext] Loading extension: gxf/libgxf_message_compositor.so
[component_container_mt-1] [INFO] [1670313007.682611882] [argus_mono]: [NitrosNode] Loading graph to the optimizer
[component_container_mt-1] [INFO] [1670313007.704270603] [argus_mono]: [NitrosNode] Running optimization
[component_container_mt-1] [INFO] [1670313007.770941913] [argus_mono]: [NitrosNode] Obtaining graph IO group info from the optimizer
[component_container_mt-1] [INFO] [1670313007.775147922] [argus_mono]: [NitrosNode] Creating negotiated publishers/subscribers
[component_container_mt-1] [INFO] [1670313007.791643284] [argus_mono]: [NitrosNode] Starting negotiation...
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/argus_mono' in container '/argus_mono_container'
[component_container_mt-1] [INFO] [1670313008.795292529] [argus_mono]: [NitrosNode] Starting post negotiation setup
[component_container_mt-1] [INFO] [1670313008.795539411] [argus_mono]: [NitrosNode] Getting data format negotiation results
[component_container_mt-1] [INFO] [1670313008.795643123] [argus_mono]: [NitrosPublisher] Negotiation failed
[component_container_mt-1] [INFO] [1670313008.795717780] [argus_mono]: [NitrosPublisher] Use only the compatible publisher: topic_name="/left/image_raw", data_format="nitros_image_rgb8"
[component_container_mt-1] [INFO] [1670313008.795806804] [argus_mono]: [NitrosPublisher] Negotiation failed
[component_container_mt-1] [INFO] [1670313008.795862421] [argus_mono]: [NitrosPublisher] Use only the compatible publisher: topic_name="/left/camerainfo", data_format="nitros_camera_info"
[component_container_mt-1] [INFO] [1670313008.796011509] [argus_mono]: [NitrosNode] Exporting the final graph based on the negotiation results
[component_container_mt-1] [INFO] [1670313008.822104273] [argus_mono]: [NitrosNode] Wrote the final top level YAML graph to "/workspaces/isaac_ros-dev/install/isaac_ros_argus_camera/share/isaac_ros_argus_camera/SWTUDGPKSZ.yaml"
[... cut ...]
[component_container_mt-1] |                          SWTUDGPKSZ_converter_left |   14881 |     2.74 |     3.50 |   493.27 |    9.6 % |   47881.8 |   221727001126 |   392260266951 |   499860250491 |
[component_container_mt-1] |                            SWTUDGPKSZ_argus_camera |   14949 |     2.28 |     3.51 |   149.44 |    8.7 % |   43524.8 |   220326933580 |   390026958519 |   499860250491 |
[component_container_mt-1] |==================================================================================================================================================================|
[INFO] [component_container_mt-1]: process has finished cleanly [pid 66891]

hemalshahNV, 2023-01-31:
The warning is because Argus was unable to read a camera calibration from the camera's built-in EEPROM but your URL should override that anyway. As far as the camera frame rate is concerned, what is your system state when running this test application (running top or jtop)? Are CPUs or GPUs pegged during this application? We're working on reproducing this issue with an IMX219 and Isaac ROS Argus Camera.

Lanzo98, 2023-02-02:
I have the same problem. I'm on a Xavier nx with IMX219 and from `ros2 topic hz /left/image_raw` I'm getting 10 fps. I have set mode to 4 and framerate to 60 in [these lines](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera/blob/main/isaac_ros_argus_camera/config/argus_camera_mono_node.yaml#L69-L72). Here some screenshots of the hz and resources usage.
[image]
[image]

ssseeex, 2023-08-22:
Do you solve the problem,?I have the same problem with Hawk camera

WangGangUCAS, 2024-05-22:
"Received an image, do you need to modify the isaac_ros_argus_camera code?"

HJGrant, 2024-07-26:
I have the same problem. I set my camera mode to 1920x1080 and 60fps but I am still only getting around <10 fps. Has anyone found a solutions to this problem ?