NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #16: Repeated "Failed to get CaptureSession interface"

cospan, 2023-03-27:
Hi,

When running a modified mono-sensor capture pipeline

```
ros2 launch isaac_ros_argus_camera isaac_ros_argus_camera_mono.launch.py
```

with the following `isaac_ros_argus_camera_mono.launch.py` file:

```
# SPDX-FileCopyrightText: NVIDIA CORPORATION & AFFILIATES
# Copyright (c) 2021-2022 NVIDIA CORPORATION & AFFILIATES. All rights reserved.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
# http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# SPDX-License-Identifier: Apache-2.0

import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    argus_mono_node = ComposableNode(
        name='argus_mono',
        package='isaac_ros_argus_camera',
        plugin='nvidia::isaac_ros::argus::ArgusMonoNode',
        parameters=[{
            'camera_id': 0,
            'module_id': 0,
            'mode': 0,
            'camera_info_url': 'file:///workspaces/isaac_ros-dev/src/camera-calibration/left_image_1232_1038/ost.ini'
        }],

    )
    argus_resize_node = ComposableNode(
        name='argus_resize_node',
        package='isaac_ros_image_proc',
        plugin='nvidia::isaac_ros::image_proc::ResizeNode',
        parameters=[{
            'width': 1920,
            'height': 1080,
            'backend':"CUDA"
        }],
        remappings=[
            ('image_raw', 'left/image_raw'),
            ('camera_info', 'left/camerainfo'),
            ('image_rect', 'left/image_rect'),
            ('camera_info_rect', 'left/camera_info_rect')
        ]
    )
    argus_rectify_node = ComposableNode(
        name='argus_rectify_node',
        package='isaac_ros_image_proc',
        plugin='nvidia::isaac_ros::image_proc::RectifyNode',
        parameters=[{
            'output_width': 640,
            'output_height': 480,
        }],
        remappings=[
            ('image_raw', 'left/image_raw'),
            ('camera_info', 'left/camerainfo'),
            ('image_rect', 'left/image_rect'),
            ('camera_info_rect', 'left/camera_info_rect')
        ]
    )

    argus_mono_container = ComposableNodeContainer(
            name='argus_mono_container',
            package='rclcpp_components',
            executable='component_container_mt',
            #composable_node_descriptions=[argus_mono_node],
            composable_node_descriptions=[argus_mono_node, argus_rectify_node],
            namespace='',
            output='screen',
            arguments=['--ros-args', '--log-level', 'info'],
        )
    return launch.LaunchDescription([argus_mono_container])
```

I receive the following
```
(cut for brevity, full application log dump attached as `post_crash_run.txt`)
...
[component_container_mt-1] [INFO] [1679920061.978564315] [argus_rectify_node]: [NitrosContext] Initializing applicaiton...
[component_container_mt-1] 2023-03-27 12:27:42.013 ERROR extensions/hawk/argus_camera.cpp@364: Failed to get CaptureSession interface
[component_container_mt-1] 2023-03-27 12:27:42.013 WARN  gxf/std/greedy_scheduler.cpp@221: Error while executing entity 8 named 'MCIWSCZCKX_argus_camera': GXF_FAILURE
[component_container_mt-1] 2023-03-27 12:27:42.020 ERROR gxf/std/entity_executor.cpp@200: Entity with 31 not found!
[component_container_mt-1] [ERROR] [1679920062.020878959] [argus_mono]: [NitrosPublisher] Vault ("vault_left_image/vault", eid=31) was stopped. The graph may have been terminated due to an error.
[component_container_mt-1] [INFO] [1679920062.024064713] [argus_rectify_node]: [NitrosContext] Running appliation...
[component_container_mt-1] terminate called after throwing an instance of 'std::runtime_error'
[component_container_mt-1]   what():  [NitrosPublisher] Vault ("vault_left_image/vault", eid=31) was stopped. The graph may have been terminated due to an error.
[ERROR] [component_container_mt-1]: process has died [pid 7249, exit code -6, cmd '/opt/ros/humble/install/lib/rclcpp_components/component_container_mt --ros-args --log-level info --ros-args -r __node:=argus_mono_container -r __ns:=/'].
```

If I try and restart the application it does the same thing, even if I try and run something outside of the ROS Docker image such as `argus_camera` I receive the same error.

* I've attempted to restart the `argus-daemon` but that doesn't work.
* If I perform a hard reset everything is fine

This is the second time this has happened. I think this occurs when I cancel a run early. I had the presence of mind to capture the log of the previous capture (`pre_crash_run.txt). My guess is that I canceled the run during an initialization period but obviously I can't look too closely into the code.

It's not a show-stopper but would be nice not to worry about this.

 
[post_crash_run.txt](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera/files/11079025/post_crash_run.txt)
[pre_crash_run.txt](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera/files/11079026/pre_crash_run.txt)

[pre_crash_run.txt](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera/files/11079027/pre_crash_run.txt)

hemalshahNV, 2023-03-29:
We have run into this before and it usually indicates that the argus daemon has gotten into a bad state. We have a line in our troubleshooting for it [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera#symptoms) but we don't have a solid fix for this yet.

jaiveersinghNV, 2023-04-10:
Could you please try this again with the latest [Isaac ROS DP3 release](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera/releases/tag/v0.30.0-dp)?

Also, could you please provide some additional information about your system:

- Which Jetson board are you using? 
- What version of JetPack is installed?
- Which camera are you using?

cospan, 2023-04-11:
**System Information:**

* Module: Xavier NX
* Jetpack Version: The output of `head -n 1 /etc/nv_tegra_release`: `# R35 (release), REVISION: 1.0, GCID: 31346300, BOARD: t186ref, EABI: aarch64, DATE: Thu Aug 25 18:41:45 UTC 2022 ` (Is this sufficient?)
* Board: We built a custom board that can interface with three Leopard Imaging compatible ipex connectors
* Sensor: [IMX265](https://www.leopardimaging.com/wp-content/uploads/LI-IMX265-MIPI-M12_Datasheet.pdf)
  * They have a custom driver that can be built for the kernel

**What I did:**
* I pulled commit `7f44d802397c0331c3eac9271c0de67540357ab4` (Merge pull request #18 from NVIDIA-ISAAC-ROS/release-dp3 )
* applied my changes to the `isaac_ros_argus_camera_mono.launch.py` and built with `colcon`
* Ran the launch file:
```
ros2 launch isaac_ros_argus_camera isaac_ros_argus_camera_mono.launch.py
```

**Results**

No issues! Great! I restarted a few times. I was quite abusive too. I canceled while it was initializing and had no problems.

If anything happens I'll re-open this issue.

closed 2023-04-11 by cospan