NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #17: LI-AR0234CS-STEREO-GMSL2 Failed to get Left IOutputStreamSettings interface

Vulcan-YJX, 2023-04-03:
`[component_container_mt-1] [1;31m2023-04-03 15:28:31.511 ERROR extensions/hawk/argus_camera.cpp@383: Failed to get Left IOutputStreamSettings interface
[component_container_mt-1] [1;31m2023-04-03 15:28:31.511 ERROR gxf/std/entity_executor.cpp@200: Entity with 48 not found!`

- camera : LI-AR0234CS-STEREO-GMSL2
- cmd : ros2 launch isaac_ros_argus_camera isaac_ros_argus_camera_stereo.launch.py

get error！

jaiveersinghNV, 2023-04-10:
Could you please provide the following information to help us debug your issue?

- Which JetPack version
- Which specific Jetson board
- Which version of the HAWK drivers are installed

Vulcan-YJX, 2023-04-11:
The above issue uses libargus from Jetpack 5.0.1 and 2022-10-19 versions. 
Now I have upgrade my prj but have a new problem. 

- Version : jetpack 5.1.1
- HardWare : Nvidia orin agx 32G
- HAWK drivers: [35.2_GA](https://github.com/LPXingxing/LI-AR0234CS-STEREO-GMSL2/tree/main/35.2_GA)

I have been able to obtain the camera's image through nvgstcapture and the camera is working properly. But I received the following error report.

- WARN  gxf/std/program.cpp@456: No system specified. Nothing to do

- [ERROR] [1681182269.461670754] [argus_stereo]: [NitrosContext] dlopen failed when opening "/home/nvidia/workspace/hawk_cam/install/isaac_ros_gxf/share/isaac_ros_gxf/gxf/lib/libgxf_argus.so": desktop-shell.so: cannot open shared object file: No such file or directory
-  ERROR gxf/std/program.cpp@475: Attempted interrupting when not running (state=0d).
- ERROR gxf/core/runtime.cpp@1243: Graph interrupt failed with error: GXF_INVALID_EXECUTION_SEQUENCE
- Component constructor threw an exception: [NitrosNode] loadExtensions Error: GXF_FAILURE
- Failed to load node 'argus_stereo' of type 'nvidia::isaac_ros::argus::ArgusStereoNode' in container '/argus_stereo_container': Component constructor threw an exception: [NitrosNode] loadExtensions Error: GXF_FAILURE

hemalshahNV, 2023-04-21:
Are you running within the Isaac ROS Dev container by any chance? You'll need `/usr/lib/aarch64-linux-gnu/tegra/weston` in your LD_LIBRARY_PATH as we configure for you in Dockerfile.aarch64 [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/88023df7376ac02421302a61349e439165557778/docker/Dockerfile.aarch64#L142).

Vulcan-YJX, 2023-04-26:
Thanks . it works.