NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #20: dlopen failed when opening libgxf_argus.so

ikhann, 2023-05-11:
Hi,

My System:
Jetson Orin NX and IMX219 camera module (Raspberry Camera v2)
Ubuntu 20.04
Jetpack 5.1

I attempted to replicate the example provided in the [Quickstart](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera#quickstart) guide, but encountered some errors:

```
admin@tegra-ubuntu:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_argus_camera isaac_ros_argus_camera_mono.launch.py
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2023-05-11-14-35-06-531736-tegra-ubuntu-11780
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [11791]
[component_container_mt-1] [INFO] [1683815706.998143565] [argus_mono_container]: Load Library: /workspaces/isaac_ros-dev/install/isaac_ros_argus_camera/lib/libmono_node.so
[component_container_mt-1] [INFO] [1683815707.087443605] [argus_mono_container]: Found class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::argus::ArgusMonoNode>
[component_container_mt-1] [INFO] [1683815707.087597881] [argus_mono_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nvidia::isaac_ros::argus::ArgusMonoNode>
[component_container_mt-1] [INFO] [1683815707.093136158] [NitrosContext]: [NitrosContext] Creating a new shared context
[component_container_mt-1] [INFO] [1683815707.093317187] [argus_mono]: [NitrosNode] Initializing NitrosNode
[component_container_mt-1] [INFO] [1683815707.094419709] [NitrosContext]: [NitrosContext] Loading extension: gxf/lib/std/libgxf_std.so
[component_container_mt-1] [INFO] [1683815707.099020652] [NitrosContext]: [NitrosContext] Loading extension: gxf/lib/libgxf_gxf_helpers.so
[component_container_mt-1] [INFO] [1683815707.102178744] [NitrosContext]: [NitrosContext] Loading extension: gxf/lib/libgxf_sight.so
[component_container_mt-1] [INFO] [1683815707.105510697] [NitrosContext]: [NitrosContext] Loading extension: gxf/lib/libgxf_atlas.so
[component_container_mt-1] [INFO] [1683815707.110105847] [NitrosContext]: [NitrosContext] Loading application: '/workspaces/isaac_ros-dev/install/isaac_ros_nitros/share/isaac_ros_nitros/config/type_adapter_nitros_context_graph.yaml'
[component_container_mt-1] [INFO] [1683815707.111268371] [NitrosContext]: [NitrosContext] Initializing application...
[component_container_mt-1] [INFO] [1683815707.113218050] [NitrosContext]: [NitrosContext] Running application...
[component_container_mt-1] 2023-05-11 14:35:07.113 WARN  gxf/std/program.cpp@456: No system specified. Nothing to do
[component_container_mt-1] [INFO] [1683815707.118107128] [argus_mono]: [NitrosNode] Starting NitrosNode
[component_container_mt-1] [INFO] [1683815707.118160794] [argus_mono]: [NitrosNode] Loading built-in preset extension specs
[component_container_mt-1] [INFO] [1683815707.120922588] [argus_mono]: [NitrosNode] Loading built-in extension specs
[component_container_mt-1] [INFO] [1683815707.120981886] [argus_mono]: [NitrosNode] Loading preset extension specs
[component_container_mt-1] [INFO] [1683815707.121802897] [argus_mono]: [NitrosNode] Loading extension specs
[component_container_mt-1] [INFO] [1683815707.121839346] [argus_mono]: [NitrosNode] Loading generator rules
[component_container_mt-1] [INFO] [1683815707.122156186] [argus_mono]: [NitrosNode] Loading extensions
[component_container_mt-1] [INFO] [1683815707.122751016] [argus_mono]: [NitrosContext] Loading extension: gxf/lib/cuda/libgxf_cuda.so
[component_container_mt-1] [INFO] [1683815707.125498282] [argus_mono]: [NitrosContext] Loading extension: gxf/lib/serialization/libgxf_serialization.so
[component_container_mt-1] [INFO] [1683815707.129580525] [argus_mono]: [NitrosContext] Loading extension: gxf/lib/libgxf_isaac_messages.so
[component_container_mt-1] [INFO] [1683815707.134171739] [argus_mono]: [NitrosContext] Loading extension: gxf/lib/multimedia/libgxf_multimedia.so
[component_container_mt-1] [INFO] [1683815707.134925678] [argus_mono]: [NitrosContext] Loading extension: gxf/lib/image_proc/libgxf_tensorops.so
[component_container_mt-1] [INFO] [1683815707.139383257] [argus_mono]: [NitrosContext] Loading extension: gxf/lib/libgxf_argus.so
[component_container_mt-1] [ERROR] [1683815707.152655865] [argus_mono]: [NitrosContext] dlopen failed when opening "/workspaces/isaac_ros-dev/install/isaac_ros_gxf/share/isaac_ros_gxf/gxf/lib/libgxf_argus.so": libnvidia-kms.so.35.3.1: cannot open shared object file: No such file or directory
[component_container_mt-1] [ERROR] [1683815707.152752155] [argus_mono]: [NitrosNode] loadExtensions Error: GXF_FAILURE
[component_container_mt-1] [INFO] [1683815707.153733715] [argus_mono]: [NitrosNode] Terminating the running application
[component_container_mt-1] [INFO] [1683815707.153778548] [argus_mono]: [NitrosContext] Interrupting GXF...
[component_container_mt-1] 2023-05-11 14:35:07.153 ERROR gxf/std/program.cpp@475: Attempted interrupting when not running (state=0d).
[component_container_mt-1] 2023-05-11 14:35:07.153 ERROR gxf/core/runtime.cpp@1243: Graph interrupt failed with error: GXF_INVALID_EXECUTION_SEQUENCE
[component_container_mt-1] [ERROR] [1683815707.153839605] [argus_mono]: [NitrosContext] GxfGraphInterrupt Error: GXF_INVALID_EXECUTION_SEQUENCE
[component_container_mt-1] [INFO] [1683815707.153852054] [argus_mono]: [NitrosContext] Waiting on GXF...
[component_container_mt-1] [INFO] [1683815707.153872022] [argus_mono]: [NitrosContext] Deinitializing...
[component_container_mt-1] [INFO] [1683815707.153892183] [argus_mono]: [NitrosContext] Destroying context
[component_container_mt-1] [INFO] [1683815707.154637641] [argus_mono]: [NitrosNode] Application termination done
[component_container_mt-1] [ERROR] [1683815707.158105660] [argus_mono_container]: Component constructor threw an exception: [NitrosNode] loadExtensions Error: GXF_FAILURE
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'argus_mono' of type 'nvidia::isaac_ros::argus::ArgusMonoNode' in container '/argus_mono_container': Component constructor threw an exception: [NitrosNode] loadExtensions Error: GXF_FAILURE
```
Could you please help me to solve this issue?
Thanks in advance.

ikhann, 2023-05-11:
I checked if the library is present on my system:

```
admin@tegra-ubuntu:/workspaces/isaac_ros-dev$ ldconfig -p | grep libnvidia-kms.so
	libnvidia-kms.so.35.2.1 (libc6,AArch64) => /usr/lib/aarch64-linux-gnu/tegra/libnvidia-kms.so.35.2.1
	libnvidia-kms.so (libc6,AArch64) => /usr/lib/aarch64-linux-gnu/tegra/libnvidia-kms.so
```

So, it looks like I have the older Jetpack version 35.1 instead of 35.1.1.

jaiveersinghNV, 2023-05-15:
Yes, as specified in our [supported platforms](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera#supported-platforms) section, we require the latest JetPack 35.1.1 release for Isaac ROS Argus Camera.

Could you try upgrading to the latest version, either by reflashing or by using the steps [here](https://docs.nvidia.com/jetson/jetpack/install-jetpack/index.html#upgrade-jetpack), and see if that works for you?

closed 2023-05-27 by ikhann