NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #21: calibration url distortion type detection issue

chengu1000, 2023-05-24:
currently, we are using rasberry pi IMX477 camera module equipped with 3.9mm F2.8 lens equipped. The exact product is https://vctec.co.kr/product/detail.html?product_no=17791&cate_no=145
because it didn't publish camerainfo by its own driver, we have made calibration yaml & ini files, to publish camerainfo. while proceeding, we met the error 
yaml-cpp: bad conversion in line 4~

so we decided to use ini file, and we found out that the node arguscamera sees the url and gets the calibration data

but we got calibration data as 'plumb_bob' distortion model, but if we echo the camerainfo publishing, it shows that distortion type is 'rational_polynomial', resulting in 3 more garbage data in the distortion parameter in the message

so we tried to put 

camera_info_->distortion_model = "plumb_bob" 

in between line 86 & 87 in 'argus_camera_mono_node.cpp', but it doesn't change the output message's distortion_model

how can we make the node to recognize the distortion model as 'plumb_bob'?

we are using the same container you've uploaded in the tutorial, and using Jetson Xavier NX, and tried to update/lower the version of yaml-cpp, but it didn't resolve the problem related in yaml file..

we upload the zip file containing the yaml and ini file for your support.
[camera_infos.zip](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera/files/11553695/camera_infos.zip)

jaiveersinghNV, 2023-05-31:
Thank you for reporting this issue. We are looking into it and will keep you updated.

jaiveersinghNV, 2023-06-05:
Hi @chengu1000 - we've just merged #23 that we think will fix your bug. Could you try pulling the latest and see if this works for you?

closed 2023-07-26 by jaiveersinghNV