NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #32: Connecting with  multiple monocular cameras

oncepursuit, 2023-10-26:
Hello
I can connect a single monocular camera successfully , but I want to connect multiple monocular cameras. 
How can I do that ? 
 Does this code support connecting multiple monocular cameras at the same time? 
Thank you.

jaiveersinghNV, 2023-10-30:
Assuming you've correctly wired the monocular cameras to the Jetson using the appropriate GMSL expansion boards, you should be able to run multiple instances of the `isaac_ros_argus_camera_node` with the appropriate settings for [`device_id` and `module_id`](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_argus_camera/isaac_ros_argus_camera/index.html#ros-parameters).

jaiveersinghNV, 2023-12-04:
Could you confirm that you are running on the latest [Isaac ROS 2.1 release](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera/releases/tag/v2.1.0) of Isaac ROS Argus Camera? Multi-instance support for Argus was a feature that was added recently. If you're running an older version of Isaac ROS, please update to the latest and try again.

WangGangUCAS, 2024-05-22:
"Received an image, do you need to modify the isaac_ros_argus_camera code?"