NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #33: Low FPS IMX477 on Jetson Orin Nano Dev Kit

furkansariyildiz, 2023-11-03:
Hello Everyone, I am using **isaac_ros_argus_camera** package to get image via using IMX477 Arducam and Jetson Orin Nano. When I run **ros2 launch isaac_ros_argus_camera isaac_ros_argus_camera_mono.launch.py** command, it works but fps is very low. Response of **ros2 topic hz /left/image_raw is 10.0 hz.**
By the way I am trying to change mode and framerate from argus_camera_mono_node.yaml but it is not changing and launch command returns **Failed to get calibration data from Argus!**

Thanks for everything!

jaiveersinghNV, 2023-11-06:
Could you confirm that you're running your Jetson in the maximum power mode? See [this comment](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_argus_camera/issues/28#issuecomment-1643886413) for more information.

Please note that Isaac ROS Argus Camera is meant to work with specific cameras whose vendors have provided camera drivers that are compatible with [NVIDIA's `libArgus`](https://docs.nvidia.com/jetson/l4t-multimedia/group__LibargusAPI.html). The list of NVIDIA-verified compatible cameras is available [here](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_argus_camera/index.html#reference-cameras).

furkansariyildiz, 2023-11-07:
Thanks for your response @jaiveersinghNV,

I am using 15W Power Mode. By the way, the camera which I try to use is IMX477 and its name is on compatible camera list.

jaiveersinghNV, 2023-11-08:
The Sony IMX477 is a popular image sensor that many different camera vendors have used as a component of their complete camera solutions. The compatibility chart for Isaac ROS Argus Camera specifically lists Leopard Imaging's IMX477-based monocular camera, not Arducam's. Leopard Imaging has worked with NVIDIA to produce a special Argus-compatible device driver. Even though your camera uses the same *image sensor* as one that is supported, there is no Arducam-provided Argus-compatible driver, and thus your camera will not work with Isaac ROS Argus Camera.

If you are interested in using Isaac ROS Argus Camera with a different camera, please contact your camera's vendor and ask them to provide a driver that is compatible with NVIDIA's Argus library.

Of course, you can always use a standard non-Argus camera driver to extract images from your Arducam camera and publish them as ROS Image messages. While there may be a performance penalty compared against an Argus-compatible camera + driver, you will still be able to use other nodes in the Isaac ROS suite of packages.

furkansariyildiz, 2023-11-08:
Thank you @jaiveersinghNV,

I will try again with **sudo /usr/bin/jetson_clocks** command again. If it will not solve my problem, I'll try to use nvargus and get image via using OpenCV libraries. Have a nice day!

closed 2023-11-08 by furkansariyildiz