NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #36: wrong cudart on jetpack 6

muttistefano, 2024-01-03:
Hi all,
i am trying this package on Jetpack 6 and even thou it builds, this error pops up when I run the mono launcher:

```
[INFO] [launch]: All log files can be found below /home/grace/.ros/log/2024-01-03-13-52-12-969463-grace-desktop-8615
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [8640]
[component_container_mt-1] [INFO] [1704289933.703623279] [argus_mono_container]: Load Library: /home/grace/ros_ws/install/isaac_ros_argus_camera/lib/libmono_node.so
[component_container_mt-1] [ERROR] [1704289933.733278228] [argus_mono_container]: Failed to load library: Could not load library dlopen error: libcudart.so.11.0: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'argus_mono' of type 'nvidia::isaac_ros::argus::ArgusMonoNode' in container '/argus_mono_container': Failed to load library: Could not load library dlopen error: libcudart.so.11.0: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
```

why is clibcudart.so.11.0 linked even if jetpack 6 comes with cuda12.2 ?
Am I doing something wrong ?
Thanks

jaiveersinghNV, 2024-01-09:
The latest Isaac ROS 2.1 release only supports JetPack 5.1.2. We don't currently support the JetPack 6, which is available in Developer Preview right now.

If possible, you should downgrade to the latest stable release of JetPack 5.1.2, and then all Isaac ROS software should work as expected.

muttistefano, 2024-01-09:
Indeed by downgrading it works as expected.
Thanks

closed 2024-01-09 by muttistefano