NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #48: `sensing' camera, how to configure

zymouse, 2024-09-03:
Hi, for [sensing](https://www.sensing-world.com/en/pd.jsp?recommendFromPid=0&id=106&fromMid=1544) monocular camera, how to configure it to use `isaac_ros_argus_camera`?

[image]

I tried to run it and got the following error

[image]
Very much looking forward to your reply!

zymouse, 2024-09-04:
@hemalshahNV 
Please help me, thank you very much 🙏

sgillen, 2025-05-27:
Hello, can you also check if the argus_camera utlity works? 

it seems like the ROS level driver is getting all zero timestamps, which is causing the frame synchronization logic to drop frames.