NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #51: Failed to start camera with JetPack 6.2

StemPlusLeslie, 2025-02-03:
Hi,

We are trying to setup IsaacRos 3.2 in JetPack 6.2.
However we found that argus camera node failed to start with imx219, the process end with the message:
[ERROR] [component_container_mt-1]: process has died [pid 2310, exit code -11, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args --log-level info --ros-args -r __node:=argus_mono_container -r __ns:=/'].

We confirmed that this is not hardware problem as the same set of hardware working with JetPack 6.1, 6.0

May I know how can we fix this?

rowanborder, 2025-03-13:
I am also experiencing this issue with an IMX296 camera on ISAAC-ROS 3.2. It appears that ISAAC-ROS 3.2 expects every camera to have an IMU, as the segfault occurs in `nvidia::isaac::ArgusCamera::ArgusImpl::fillImuCalibrationData`. This is not the case for cameras like the IMX296.

Reverting to ISAAC ROS 3.1 using this fix (https://forums.developer.nvidia.com/t/argus-camera-segfault-on-jetpack-6-2/324192/9) works, but I would like to upgrade to ISAAC ROS 3.2 to use the new hardware timestamping feature.

sgillen, 2025-05-23:
Hi @StemPlusLeslie, are you able to run the argus_camera sample application? That binary should be installed with jetpack. That will help us triage the issue.

josepqp, 2025-06-20:
same problem here. Also using ConnectTech Boson Carrier Board, but I use 2 IMX565 cameras from Framos

I can use ISAAC ROS 3.1 with frame drops (like reported by other users), but I can not use ISAAC ROS 3.2

I'm using JetPack 6.2 – L4T r36.4.3

sgillen, 2025-07-11:
This is likely a dupe of #54, using that bug for the main discussion, please try the fix there!.

closed 2025-07-15 by jaiveersinghNV