NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #53: Argus Camera Segfault on Jetpack 6.2

Calvin-InDro, 2025-02-14:
This was previously working fine with Jetpack 6.0/6.1 utilizing the IMX219. The segfault presents as below:

[ERROR] [component_container_mt-1]: process has died [pid 2310, exit code -11, cmd ‘/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args --log-level info --ros-args -r __node:=argus_mono_container -r __ns:=/’]

Potentially related issue:
We are using previously functional DTOs which enabled the simultaneous use of multiple IMX219s with Jetpack 6.0/6.1. On Jetpack 6.2, we can only open one camera at a time with the nvarguscamera gst plugin.

Any insight greatly appreciated!

Infinite-Echo, 2025-02-21:
I have the same issue. Hopefully, Nvidia will fix this soon.

sgillen, 2025-05-27:
This look like a duplicate of #51, let's move discussion there.

closed 2025-07-15 by jaiveersinghNV