NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #55: Multiple Image Pipeline nodes not compatible with multiple Argus Cameras

Calvin-InDro, 2025-05-27:
Jetpack 6.0/6.1
x2 IMX219
x2 isaac_ros_argus_camera mono nodes
x2 isaac_ros_image_proc rectification nodes

While able to access any CSI connected IMX219s and stream multiple simultaneously via gst on Jetpack 6.0/6.1, opening multiple isaac_ros_argus_camera mono nodes simultaneously with different cameras results in strange behaviour from the isaac_ros_image_pipeline. When attempting to rectify the images for example, one stream (random if stream NVIDIA-ISAAC-ROS/isaac_ros_image_pipeline#1 or stream NVIDIA-ISAAC-ROS/isaac_ros_image_pipeline#2) will be rectified perfectly, while the other results in an empty image. Sometimes, however, the second stream appears to utilize nonsensical intrinsic camera parameters (while other times, rectifying completely normally). These are low-distortion lenses, and the camera calibrations are accurate, and DO result in perfectly rectified images when applied correctly (plum_bob models).

Pictured here are the rare times the second stream results in a non-empty image with nonsensical rectifications:

[image]
[image]

Any insight on this behavior would be greatly appreciate.

rowanborder, 2025-05-28:
I captured the argus logs from two attempts starting the camera driver node, `nvidia::isaac_ros::argus::ArgusMonoNode`, and the rectify node, `nvidia::isaac_ros::image_proc::RectifyNode`, - one where a black image was generated and the other where the image was correctly rectified. Unfortunately I can't notice a difference between the two.

I have also previously tried adding a delay between starting the camera driver node and the rectify node, in case there was some kind of race condition that started the rectification before the camera node was properly initialised, but this was unsuccessful.

[nvargus-daemon_rectify_correct_image.txt](https://github.com/user-attachments/files/20479620/nvargus-daemon_rectify_correct_image.txt)
[nvargus-daemon_rectify_black_image.txt](https://github.com/user-attachments/files/20479619/nvargus-daemon_rectify_black_image.txt)

sebastian-indro, 2025-06-02:
@sgillen Hi again Sean, is this the info you needed? If something is missing we can provide more
Cheers thank you for helping on this issue!

sgillen, 2025-06-04:
Arai got back to me offline, we don't see any argus level issues, and I don't see anything obvious from the ros level logs that were shared. 

So lets try to narrow this down further, is it just the rectification that has issues or is it the output from the camera node? Are any of you able to replicate this issue without the rectification node? It may also help to get more verbose logs, which we can get with:

```export GXF_LOG_LEVEL=info```

rowanborder, 2025-06-06:
I ran the tests again with `export GXF_LOG_LEVEL=info` and have attached the logs. I don't notice any difference in the additional messages though.

[isaac_ros_logs_rectify_correct_image_gxf_info.txt](https://github.com/user-attachments/files/20629327/isaac_ros_logs_rectify_correct_image_gxf_info.txt)
[isaac_ros_logs_rectify_black_image_gxf_info.txt](https://github.com/user-attachments/files/20629328/isaac_ros_logs_rectify_black_image_gxf_info.txt)

vixp24, 2025-08-27:
Has anyone successfully resolved this issue? @sgillen I can confirm that the output of the Argus node is always "working" in this failure condition, and so is the output of a downstream resize node. Keep in mind this is viewed in rqt, so it would be subscribing to the sensor_msgs/Image, not the NitrosImage that RectifyNode would be using. I've also found that when ArgusMonoNode is replaced with a gscam2 node, the RectifyNode does not exhibit this behaviour.. Any thoughts? 

I've documented my findings [here](https://forums.developer.nvidia.com/t/image-resize-and-image-rectify-publish-empty-images/298678/10?u=vpanchal1).

Thanks!

vixp24, 2025-10-29:
@sgillen Do you have an update on the status of this bug? Is it resolved in Isaac ROS 4.0?