NVIDIA-ISAAC-ROS/isaac_ros_argus_camera #57: Frame drop using hardware trigger

josepqp, 2025-07-14:
I'm using isaac ros in version 3.2 with two IMX565 from Framos, using connecttech boson board.

I have both cameras synchronized with a hardware trigger, I'm using low frame rate. fps = 1

I'm getting a lot of warning due to it:

2025-07-14 08:35:08.367 WARN  extensions/hawk/components/argus_camera.cpp@1735: Frame drop in module_id 0 camera_id 0
2025-07-14 08:35:08.379 WARN  extensions/hawk/components/argus_camera.cpp@1735: Frame drop in module_id 1 camera_id 1

In previous issues, it was suggested to try:

GXF_LOG_LEVEL=ERROR

but warning already appear. Any other solution ?

sgillen, 2025-07-16:
Just to clarify, you ran `export GXF_LOG_LEVEL=ERROR` in the same terminal that you later run the application in? Can you try a typo just to make sure that the env variable is being seen? If you try for example `export GXF_LOG_LEVEL=TYPO ` then it should error out when you run your application.

josepqp, 2025-07-18:
@sgillen it seams it was my problem. Now I don't have warnings.

closed 2025-07-18 by josepqp